BainbridgeArtisanResourceNetwork/Mark4_printer #1: BOM total build cost is bogus

wabmester, 2019-11-29:
Any reason not to use "=SUM(E2:E73)" for that field?    Also, suggest we include an "Actual" column for cost of parts purchased.

wabmester, 2019-11-30:
So.... I installed OpenOffice, forked, made some changes, and then discovered that OpenOffice can't save in the "xlsx" format.    It can save "xls" though.    What did you use to create the xlsx, Microsoft excel?

mikeabuilder, 2019-11-30:
I did use excel, based on your request. I have no major problem swapping over to open office, even though it's missing one of my favorite functions (ONERROR()).  And the root of the issue is that I deleted the subtotals out of the original spreadsheet, but didn't add a new total.

wabmester, 2019-11-30:
I'll delete the xlsx and replace with and xls that both openoffice and excel can open.    I added a subtotal column and an "actual" column to reflect what we spend.    Also fixed the total cost.

wabmester, 2019-11-30:
Pull requested.    Ball's in your court.   :)

closed 2019-11-30 by wabmester