BainbridgeArtisanResourceNetwork/Mark4_printer #7: Y-Rail in BOM is 400mm. Should it be 500mm?

mikeabuilder, 2019-12-08:


mikeabuilder, 2019-12-08:
The bridge part provided by Mark looks to be designed for a 500 mm rail. Our BOM has a 400mm X-Rail.  If we keep the 400mm rail, we'll need to move the mounting holes on the bridge.

mikeabuilder, 2019-12-08:
Maybe the bridge part is too wide. When I added the main coreXY plate to the CAD model, that's what it looks like.

wabmester, 2019-12-10:
See https://github.com/BainbridgeArtisanResourceNetwork/Mark4_printer/issues/4

There was an error when transcribing Mark's BOM.    That's why I ordered 500mm rails for both X and Y.

mikeabuilder, 2019-12-11:
Just added the "build volume" to the Fusion model, and I can verify that we need 500mm rails in the Y direction in order to support a 350mm Y part size and also have additional travel available for a tool changer. I'll update the BOM, and I'm closing this issue,

closed 2019-12-11 by mikeabuilder