BainbridgeArtisanResourceNetwork/Mark4_printer #8: X-axis rail: MGN 15H we purchased too wide for Mark's bridge.

mikeabuilder, 2019-12-12:
The pulley spacing on Mark's bridge is too close to support a 15mm rail. I've switched to an MGN 12H rail in the Fusion model. We already bought a pair of 15mm rails, which we can use on the Y-axis. I'll update the BOM to make the X rail 12mm.

wabmester, 2019-12-12:
Recall the discussion about MGN12H vs MGN12C.    Verify that we don't want the smaller slide.

mikeabuilder, 2019-12-12:
12H block is about 10mm longer (in direction of travel) than the 12C, and the H is rated for much higher torque (about 150% of the C). There is a spec table in the materials folder (https://github.com/BainbridgeArtisanResourceNetwork/Mark4_printer/blob/master/Materials/linear_rail_X_axis/Similar_part_different_supplier_use%20MGN15H.jpg)  Both C and H are the same width. If the 10mm makes a difference, I'll be surprised. In any case, no purchasing until we like the Fusion design, so we'll see a problem if the "H" block is too big.

wabmester, 2019-12-12:
Jubilee went with 12C on X because it was sufficient to handle torsional load and gave a smaller footprint.    Looks like the 12H sticks out a bit in your design.    I'm sure either works fine.

mikeabuilder, 2020-03-09:
Currently running a 12C block on the proto (thanks for the loaner Doug) and have a 12C on order. When we finalize the carriage design, we will have both blocks in hand and can choose the one we like best. Closing this issue

closed 2020-03-09 by mikeabuilder